mmone/OctoprintKlipperPlugin #3: Collaborate?

shoreliner, 2018-03-31:
First, the not so much an issue part:

I reached out to you via email as well -- would love to collaborate on this project; not only have I been working on this.I also have something else in the works specifically for the extremely popular Wanhao i3 Plus/Monoprice MSP/Cocoon Create.

Now the Issue:
I am having primary install issues... even when the plugin is installed and enabled, I can't find it.   Klipper is working just fine.  It appears to be installed correctly but it is like Octoprint is filtering it out.  Glad to help trobuleshoot.

mmone, 2018-04-01:
Hi @shoreliner 

I will respond to your email later.
On your install issue: I assume you are running the most current version of octoprint and use the most current version of the plugin from github. 
The install issue is hard to diagnose without a log. If you have a standard install of octoprint the log file should be placed at "~/.octoprint/logs/octoprint.log". Please ssh into your Pi and execute "tail -F ~/.octoprint/logs/octoprint.log" this gives you a constantly updating octoprint log. Now try again to install the plugin with the Plugin Manager and check if any errors come up.

mmone, 2018-04-04:
Hi @shoreliner any news on the install issue?

shoreliner, 2018-04-05:
Hey Martin

I am going to try it again tonight.  And thanks for the ifnormation by the
way -- this latest klipper release candidate has a lot of improved hooks,
etc for the terminal and as such I think I may be able to grab some of the
data more easily with them.  I'll let you know after I try installing in a
bit.

shoreliner, 2018-04-11:
> Any news on this issue? If not I will close as I can't reproduce it.

Hey -- Apologies, haven't been very active lately.  Actually, after I moved
to the 1.3.7  RC I could no longer produce the issue.  I had a lot of
plugins installed on the prior instance, and my guess is that it was some
kind of conflict or simple browser issue that was keeping me from seeing
the Klipper plugin tab/CSS changes to the octo interface like the changes
to the 'Connection' dialog.  Anyways, this time around I will be adding
plugins one-by-one, if something conflicts I'll let you know and it should
be easier to pinpoint.

Thanks

David

mmone, 2018-04-11:
OK thanks for the update. I will close this then.

closed 2018-04-11 by mmone