mmone/OctoprintKlipperPlugin #4: Macros

ich777, 2018-04-30:
Hello,
is it possible to make the headline "Macros" invicible or a option to turn it off?
Also it would be nice if you can make something like the Babysteps as a preset so that you didn't need it to make it yourself (this is only a idea and maby not necessary)?
Regards, Christoph
[image]

mmone, 2018-05-01:
Why do you want to remove the Macro headline. To have a more compact layout?
Presets could be an interesting addition.

ich777, 2018-05-02:
@mmone Thank you for the quick reply.
Yes, exactly.
I think a option to turn the headline on and off would be a perfect solution, if thats possible and not to much work for you.
Another question, is it also possible to move the "Disconnect" button to the top bar (clicked accidentaly a few times on it)?

Btw, this plugin is super usefull, thank you for the great work and time you spent on this plugin!

mmone, 2018-08-09:
The new release optimized spacing somewhat. Closing.

closed 2018-08-09 by mmone