mmone/OctoprintKlipperPlugin #6: [Feature request] Configuration File Edit

JustAnother1, 2018-05-29:
I would like to be able to edit the Klipper configuration file from the Octoprint Web Interface.

## Use Case:
I have a delta printer running Klipper. To do the bed leveling I need to change the position of the end stop in the configuration file. If the nozzle at G1 Z0 is still too high I need to increase the position of the three end stops. if the nozzle is already touching the print bed then I need to lower the position. As all three end stop positions need to be altered every time this is a bit unpleasant.

## possible solutions:
  * configuration file editor 
  * configuration change window
  * both

One way to solve it would be a "Text editor Window" that has the contents of the configuration file and lets me edit it just like every text editor.
The other way would be a fancy window that has buttons and spinners and check boxes and whatever it needs to set all the settings possible in a configuration file. If the user then clicks save the check boxes and spinners are parsed and a configuration file is created with the configured settings. This would probably be easier for people that don't know the configuration file by hearth.
If the fancy dialog does not have all the settings (to not confuse the user) the text editor could be used as "Expert mode" view of the fancy window.

## next steps:
What would be the next steps to make this happen? What can I do to help make this happen?

mmone, 2018-05-29:
Hi,

a config file editor would be an interesting thing. As far as file permissions go it shouldn't be a problem to read/write the config file with the plugin. 

For your specific use case Klipper might already have a suitable solution though: [delta-calibration](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#delta-calibration)

A texteditor integrated into the plugin settings is certainly the most straitforward solution and could be a good testcase to get into OctoPrint plugin development. 
The documentation is quite good and comes with a tutorial: [OctoPrint documentation](http://docs.octoprint.org/en/master/plugins). 

A form based editor is definitely doable but would create quite some initial work and ongoing maintenance as the klipper interface is still evolving. One hurdle there is the ancient Bootstrap version OctoPrint uses that makes creating forms a pain.

Lust but not least I think editing the config file via SSH lets you iterate pretty fast also. If you prefer to use a none commandline Interface you can use a sftp capable file browser like [Cyberduck](https://cyberduck.io/). This allows you to edit your file locally with your preferred text editor and upload automatically when you hit save.

In conclusion: I have no immediate plans/time to implement this feature but would very much like a PR :-)

Martin

JustAnother1, 2018-05-29:
> For your specific use case Klipper might already have a suitable solution though: delta-calibration

I assume that that needs a bed level sensor, that I don't have.

> The documentation is quite good and comes with a tutorial: OctoPrint documentation.

I will look into that.

outlookhazy, 2018-07-17:
@mmone One possibility on the configuration/form front might be to only support editing of values for _existing_ keys in the config file, and potentially only certain subsets of those (i.e. calibration/adjustment values). 

My biggest annoyance with the remote text editor/ssh approach is editing the config in one window, resetting klipper in another window to apply the settings, then controlling klipper with yet another to test said settings. Potentially more than that with a slicer open etc. 

An integrated solution would be able to automatically apply a restart command after applying changes. There's also the added benefit of data validation. Once you have an "in" to the config file at the k/v pair level, implementing things like automatically applying PID calibration values is pretty trivial.

mmone, 2018-07-17:
I need no convincing that an integrated editor might be a convenient thing to have, though I personally don't seem to miss this feature as much as others do. To be quite honest it's just that other projects are more pressing/interesting at the moment. I'm still hoping that someone might be able to contribute this feature.

Maybe I should write some code overview doc to lower the entry barrier. The server/client separation might be a hurdle in the beginning. I still think though that the official plugin tutorial mentioned above is much better than what I could provide.

Your idea to limit editing to existing keys would definitely reduce the effort and the maintenance impact of a still evolving Klipper interface. It would also do same same for it's usefulness in my opinion as the first hurdle for a new user is rather to get an overview of the available options and get them right then to modify a working config.

If I could dream up a solution that would be for Klipper to provide some kind of introspection capability that would allow a client to query for available configuration keys/commands and their parameters and accepted values. I have looked briefly into what changes that would entail on the Klipper side of things and as expected these would be quite significant so I wouldn't expect Kevin to accept this as a PR at the moment.

mmone, 2018-08-09:
With [0.1.5](https://github.com/mmone/OctoprintKlipperPlugin/releases/tag/0.1.5) I've integrated a simple config editor and amongst other things cleaned up the code a bit.
Testers are very welcome. 
To update just reinstall with the plugin manager from this url: [https://github.com/mmone/OctoPrintKlipper/archive/master.zip](https://github.com/mmone/OctoPrintKlipper/archive/master.zip).
Existing settings should be automatically migrated.
Starting with this release the plugin will also inform about available updates in OctoPrint when they become available.
I will probably add at least basic syntax highlighting to the config editor soon.

Martin

outlookhazy, 2018-08-09:
@mmone Very cool, I'll try to remember to check it out this weekend (lots of other projects going on)

mmone, 2018-08-25:
Closing this as a basic config editor is now part of the plugin. Down the road there is probably gonna be a form based editor as well.

closed 2018-08-25 by mmone