mmone/OctoprintKlipperPlugin #10: Don't close window for Coordinate offset automatically

arminth, 2018-08-14:
If you key in an offset  - let's say -0.05 -  and tick the "Add to existing offset" box, the window should not close on pressing "set offset".
If it would stay open, by multiple times pressing the "set offset" it could mimic the function of babystepping, which can be very helpful for first layer adjustments.
Another idea would be having a button "set offset +" and another "set offset -". This would be even more convenient as you would choose the amount of babysteps and the rest can be done just with the two buttons.

Other than that, great job! THX!

mmone, 2018-08-14:
Thanks for the feedback. I just pushed a release that fixes this. The "+/-" style dialog might be in a future release. You would  also need a field to set the step size I think.

Octoprint should inform you that there is a new version available. If you want to get it right away goto "Software Update" in the settings and under "Advanced options" click "Force Update".

arminth, 2018-08-14:
Wow! That was fast! Thanks a lot!

arminth, 2018-08-14:
Works perfect!
Just experienced for the second time a "exceeds maximum extrusion" break while adding offsets into an ongoing print.
Don't know if it's an issue  with timing when to insert the offset. Also don't know who is inserting - the firmware or the plugin.
Cheers
Armin

mmone, 2018-08-14:
Hi Armin
that is a problem on the firmware side. Seems others are having this problem too, you can search the Klipper issues find several mentions of this error: [max extrusion error](https://github.com/KevinOConnor/klipper/issues?utf8=%E2%9C%93&q=maximum+extrusion)

closed 2018-08-19 by mmone