mmone/OctoprintKlipperPlugin #11: empty settings and macros

St007, 2018-08-14:
Installed plugin on Octoprint Version 1.3.9, Klipper Firmware version: v0.6.0-339-g3f59c3d
Running on latest Armbian.
After install I can't open klipper config (it is just empty), all settings in plugin empty, default emergency stop macro empty too. The only thing working  - panel buttons like "get status", "firmware restart". However Performance Graph is empty too (no files in dropdown file selector)

mmone, 2018-08-14:
I currently test only on octopi so some paths might be different there.

> After install I can't open klipper config (it is just empty)

You can change the path the plugin is looking for the "printer.cfg" in the settings.

> all settings in plugin empty

That I can't explain atm. The emergency macro exists but the "GCODE" field is empty?
Can you create new macros and are they persistent if you save the settings?

> However Performance Graph is empty too (no files in dropdown file selector)

The plugin looks for logfiles in /tmp/ namped klippy.log* are they also located there for your install?

In any case it would help if you could post an octoprint.log. On a standard octopi installation this is located at "/home/pi/.octoprint/logs/octoprint.log"

arminth, 2018-08-14:
In a standard install, the klippy.log ist found here: /tmp/klippy.log
I also don't get no filechoice for the perfgraph!
Cheers
Armin

St007, 2018-08-14:
more details:
1. I used plugin on another armbian with more old Octoprint, Klipper and plugin version and it works there out the box, till now.
2. I have added path to printer.cfg without any effect
3. Emergency macro is empty, there is no any macros after fresh install
4. Log file on my machine is /tmp/klippy.log
log file attached
[octoprint.log](https://github.com/mmone/OctoprintKlipperPlugin/files/2288509/octoprint.log)

mmone, 2018-08-14:
> In a standard install, the klippy.log ist found here: /tmp/klippy.log
> I also don't get no filechoice for the perfgraph!

That is interesting. Could you verify that you have logs in /tmp/ at the moment. 
If so reload you octoprint in the browser as the listing is only done once when loading the page at the moment

St007, 2018-08-14:
Yes, I can add the path, but it just not works ...not sure if it is hard to detect current user's home path and look for printer.cfg there instead of trying to load  '/home/pi/printer.cfg'

mmone, 2018-08-14:
Yes that would be possible an absolute path seemed more flexible. Last thing you could try after checking whether the path displayed in the UI is correct is to restart Octoprint as the settings are only read on startup at the moment.

St007, 2018-08-14:
Yes, the path in settings exists after Octoprint restart, that was the first I tried to do :)

mmone, 2018-08-14:
I'm trying to reproduce the issue. What is you actual path?

St007, 2018-08-14:
actual path /home/slv/printer.cfg
I guess it is in octoprint log too

mmone, 2018-08-14:
If you still willing to debug some more you could check your Octoprint config file. 
For me this is located at ~/.octoprint/config.yaml
That is where the plugin stores its settings. The should be an entry:

```
  klipper:
    _config_version: 2
    configuration:
      path: /home/slv/printer.cfg
```

St007, 2018-08-14:
```
  klipper:
    _config_version: 2
  softwareupdate:
    _config_version: 6
    check_providers:
      fanspeedslider: fanspeedslider
      klipper: klipper
printerProfiles:
  default: _default
serial:
  additionalPorts:
  - /tmp/printer
  autoconnect: true
  disconnectOnErrors: false
  port: /tmp/printer
server:
  firstRun: false
  onlineCheck:
    enabled: true
  pluginBlacklist:
    enabled: true
  secretKey: F0yaRdQAGtdQmY7DCRop9dJNK2qmZ4lO
  seenWizards:
    corewizard: 3
    cura: null
```

mmone, 2018-08-14:
OK. I think I can reproduce the issue know. It's a bug. As a quick fix you could manually add the correct path to the config.yaml like that then the rest of the default config should load as well. I will provide a bugfix release soon.

```
  klipper:
    _config_version: 2
    configuration:
      path: /home/slv/printer.cfg
```

mmone, 2018-08-14:
The release will include the possibility to configure the path for the klipper logfiles. Where are they located for your install?

St007, 2018-08-14:
klipper log file is in its default location /tmp/klippy.log

mmone, 2018-08-14:
OK. Thanks for the debug session.

St007, 2018-08-14:
Thank you for great plugin!

closed 2018-08-14 by St007

arminth, 2018-08-17:
Sorry for late reply, on business trip.
I installed 0.2.2 today (after upgrading pip). I have the klippy.log in /tmp . Still, I don't get any file displayed in the choice for the log file. I restarted octoprint. No change.
[image]

Cheers
Armin

mmone, 2018-08-17:
Please post a octoprint.log otherwise I can't really diagnose the problem.

arminth, 2018-08-17:
[octoprint.log](https://github.com/mmone/OctoprintKlipperPlugin/files/2298487/octoprint.log)
This is my last log.

mmone, 2018-08-17:
Hi Armin,
I can't find anything at the moment. The log is very long and I dont know when the dialog was opened. It would be great if you could delete you local log file so that a new one is created by octoprint. Then please restart octoprint, open the graph dialog and hit the refresh button for the file listing. Then post this log.
Is everything else working? Especially can you view the Klipper Configuration in the settings dialog?

arminth, 2018-08-17:
[octoprint(1).log](https://github.com/mmone/OctoprintKlipperPlugin/files/2298646/octoprint.1.log)
Done!
Everything else including editing Klipper configuration works fine.

mmone, 2018-08-17:
Hmm. Looks clean.

Given you have shell access to the pi we can directly execute the commands I use to list the log files.
In a shell execute the following:

```
cd                                          # to go to the user directory
source oprint/bin/activate   # activate the virtual environment used by octoprint
python                                  # launch a python console
import glob
glob.glob("/tmp/*.log*")
```
The last command should list all klippy logfiles in the directory.

arminth, 2018-08-17:
`2018-08-17 20:56:49,291 - tornado.access - WARNING - 403 POST /api/plugin/klipper (::ffff:192.168.1.40) 45.13ms
2018-08-17 20:56:55,689 - tornado.access - WARNING - 403 POST /api/plugin/klipper (::ffff:192.168.1.40) 48.96ms
2018-08-17 20:56:57,211 - tornado.access - WARNING - 403 POST /api/plugin/klipper (::ffff:192.168.1.40) 44.05ms
2018-08-17 20:56:58,707 - tornado.access - WARNING - 403 POST /api/plugin/klipper (::ffff:192.168.1.40) 45.99ms`

Looks like the webserver denies access. Just to check, I tried both nodes on octoprint: access control enabled and disabled. No change...

mmone, 2018-08-17:
Hm good catch missed that one :-) Is the octoprint user you have currently logged in "Admin"?

arminth, 2018-08-17:
ls -l /tmp/*log.*  doesn't work!

ls: cannot access '/tmp/*log.*': No such file or directory

if I change it to 
ls -l /tmp/*log

I get
-rw-r--r-- 1 pi pi 8596690 Aug 17 21:03 klippy.log

cd
pi@octopi:~ $ source oprint/bin/activate
(oprint) pi@octopi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import glob
>>> glob.glob("/tmp/*log")
['/tmp/klippy.log']

So maybe typo in the wildcard-selection?

arminth, 2018-08-17:
Using Firefox btw.

mmone, 2018-08-17:
I wrote "ls -l /tmp/*log.*"  and glob.glob("/tmp/*.log*")

arminth, 2018-08-17:
Aber an dem liegt's wohl eher nicht! Ich geh mal ins Bett! Bis morgen!

mmone, 2018-08-17:
Bis dann ich denke mal nochmal nach ... :-)

arminth, 2018-08-17:
Du hast aber oben tatsächlich
`ls -l /tmp/*log.* `
geschrieben!

Guck mal!

arminth, 2018-08-17:
Kein Stress übrigens. Das Plugin läuft sonst super! Macht deutlich mehr Spaß mit Klipper jetzt!
Danke dafür!

mmone, 2018-08-18:
Mit dem neuen release 0.2.3 sollte es funktionieren.
lg
Martin

arminth, 2018-08-18:
Sehr geil! Danke Dir! Es druckt gerade - danach teste ich sofort!

arminth, 2018-08-18:
Klappt!! Sauber!
[image]