mmone/OctoprintKlipperPlugin #12: The plugin does not show up.

blaskkaffe, 2018-08-16:
Plugin does not show up after being installed. It says "Unknown" plugin installed.

Octoprint version 1.3.9 and I have tried pip versions 9.0.3 , 10 and 18 no difference.

Also the plugin does not show up in the octoprint plugin repo and must be installed from the link or from a download.

mmone, 2018-08-16:
> Plugin does not show up after being installed. It says "Unknown" plugin installed.
Have you used the Plugin Manager to install it or pip directly?

Could you please check the octoprint.log in ~/.octoprint/logs

blaskkaffe, 2018-08-16:
I have used both the Plugin Manager(tho the plugin does not show up in the list of plugins and must be installed through the link provided on github, or manually through an uploaded zip) and through the pip command, neither seems to work.

The plugin is installed and is physically inside the correct folder but does not seem to register/show up/load correctly. When the installation is finished through the plugin manager the popup message says "unknown plugin installed"

I have checked the log file (will provide it later when I have access to the computer) and it seems to have some problem loading the plugin.

mmone, 2018-08-16:
The Plugin Manager is the recommended way to install the plugin. 
Lets continue debugging when you have access to the log file.

mmone, 2018-08-16:
Give it another try installing it with the Plugin Manager from https://github.com/mmone/OctoPrintKlipper/archive/master.zip your problem is most likely fixed now.

blaskkaffe, 2018-08-16:
Tried reinstalling and now it works! Great :)

mmone, 2018-08-16:
Thanks for reporting. Will leave this issue open for a while. Others might have run into that bug as well

arminth, 2018-08-17:
When I wanted to install the plugin (from the downloaded ZIP), I also faced the issue that the plugin just disappeared. Tryying to reinstall ended in an error parsing pip output. I could fix it by upgrading the pip shipped with Octoprint to the newest version by entering 
`sudo /home/pi/oprint/bin/python -m pip install --upgrade pip`

Hope that helps somebody.

Cheers
Armin

closed 2018-08-21 by mmone