mmone/OctoprintKlipperPlugin #15: Question: Bed Leveling feature, how to use it?

micheleamerica, 2018-08-24:
Hi

Could you please explain how to use the Bed Leveling feature of this plugin? A small Howto would be nice.

I understand that I must first determine the set of point that I want to probe, no issue there.

When running the Bed Leveling is the probe used or is it a manual process? Is it possible to configure if we want to use the probe or not without touching the printer.cfg file?

And at the end of the process, probe or manual, what happens to the results? Are they automatically applied or is it necessary to edit the printer.cfg file with the results?

Thanks in advance for your help.

Michele

mmone, 2018-08-24:
The "Assisted Bed Leveling" is supposed to assist with the well known "Paper Method" where you move the print head in a cold state to several positions on the bed and adjust the print-beds leveling screws until you feel a slight resistance when moving a piece of paper below the nozzle.
It does not enable/configure any kind of software based leveling. 

The thickness of the paper is supposed to compensate for the increase in length the hotend assembly experiences because of the thermal expansion when heated. 

To use this feature you will need to adjust the "Probe Height" parameter under "OctoKlipper/Bed Leveling/" to the thickness of the paper you are using.
It is set to 5mm by default for two reasons:
- To prevent head crashes when people try it the first time. 
- To enable my modified method of leveling where I use a 5mm precision ground dowel pin with a already heated hotend instead of the paper.

Hope this helps. I will include an explanation in the next release of the plugin.

Martin

micheleamerica, 2018-08-24:
Now I understand better the purpose of this feature of the plugin.
I was with the impression that it would allow a specific grid of points to be used for an automatic bed leveling.
Thanks for clearing this misinterpretation, and for the work you put in to give us simple users the tools to make the most out of this fabulous firmware that is Klipper.
Michele

mmone, 2018-08-24:
Lets leave this open for others.