mmone/OctoprintKlipperPlugin #18: Feature request: Klipper toggle

daef, 2018-08-26:
I would like a more convenient way to switch between marlin and klipper based printers. Currently I ssh onto the pi to `sudo service klipper stop` and disable the OctoKlipper plugin in the Settings if I want to try a marlin config. Maybe a toggle in OctoKlippers connection panel that hides/shows the OctoKlipper UI and stops/starts the klipper service.

mmone, 2018-08-26:
You can disable the simplification of OctoPrints connection panel in the plugins settings ("Settings/OctoKlipper/Basic/Replace Connection Panel") there should be no need to disable the plugin.
To control the klipper service from inside OctoPrint is not that simple as "sudo" is required. I need to check how OctoPrint handles that for the OctoPrint service.

fritzw, 2018-11-12:
> To control the klipper service from inside OctoPrint is not that simple as "sudo" is required. I need to check how OctoPrint handles that for the OctoPrint service.

On raspbian/octopi, the user `pi` can use sudo without providing a password, so octopi just uses commands like `sudo service octoprint restart`. If you disable that, you can always configure sudo to allow specific commands without a password by adding lines like this to `/etc/sudoers`:

```
octoprint ALL=NOPASSWD:/usr/sbin/service klipper start
octoprint ALL=NOPASSWD:/usr/sbin/service klipper stop
octoprint ALL=NOPASSWD:/usr/sbin/service klipper restart
```

daef, 2018-11-14:
you could also set the SUID bit on your binary I guess...

Haffwa, 2018-11-19:
Yes, this feature would be very helpful!