mmone/OctoprintKlipperPlugin #20: Klipper Console Output does not show newlines

oderwat, 2018-08-27:
I am using DELTA_CALIBRATE with some macros (using my phone as 'remote control' standing before the printer). Before I ask to add it as a feature that the plugin updates the configuration automatically I would be already pretty happy if it would show the newlines in the output. I guess this applies to other commands (mesh points?) too.

[image]

oderwat, 2018-08-27:
There are for sure newlines there normally. You probably need to convert those to `<br>` for the HTML output.

mmone, 2018-08-27:
Thanks for the hint. This is now fixed in _master_.

closed 2018-08-27 by oderwat