mmone/OctoprintKlipperPlugin #21: Feature Request

dushyantahuja, 2018-08-27:
About your environment:
 - Octopi 0.14.0
 - Octoprint version 1.3.9

Can you please add an option to hide the status messages on the menu bar. Many times the messages are old as klipper. e.g. klipper errored out saying "Printer is not ready". Now, even when the printer is ready, the menu bar keeps showing "Printer is not ready" - and to clean that up, I have to issue the status command. 

It would be great to have an option to hide the status messages.
Thanks.

oderwat, 2018-08-27:
I would already be satisfied if it would get cleared when clearing the klipper log.

mmone, 2018-08-28:
I talked to Keving about this. It's in the backlog to add a better status reporting capability to Klipper to enable a more current status display.

dushyantahuja, 2018-08-28:
That's great - however an option to simply disable this would be great. I
don't like the menu bar to be cluttered.

mmone, 2018-08-28:
Fair enough will be part of the next release.

dushyantahuja, 2018-08-28:
Thanks a lot. This is amazing.

mmone, 2018-08-28:
With a improved status reporting it can be reduced to a simple "LED" that signals the health state of the MCU and the host process until then it's indeed ugly.