mmone/OctoprintKlipperPlugin #22: Navbar show klipper errors

githubmath, 2018-09-03:
Cura add always the gcode command M205.

Klipper doesn't support this command and wrote to the terminal "Unknown command:"M205". That is currently no problem, klipper ignore this.

But OctoprintKlipperPlugin writes this "error" allway to the navbar. Thats looks not very nice.
Can you add a filter or an option to disable the message in the navbar?

I cannot disable the gcode command M205 in cura and klipper ignore it.

oderwat, 2018-09-03:
You could also add a dummy gcode macro to klipper to "eat" that command.

oderwat, 2018-09-03:
You could do something like this:

```
[gcode_macro m205]
gcode:
  G4
```

Not sure if G4 works .. may need `G4 P0`. Or something else which does not really do something.

githubmath, 2018-09-03:
Hey thanks. That its.

I add only this without something more:
```
[gcode_macro m205]
gcode:
```

And the unknown command error is gone. 👍 

Issue can close...

oderwat, 2018-09-04:
I think you can close the issue, don't you?

closed 2018-09-04 by githubmath

dotps1, 2018-09-05:
can you tell me what config file that goes in?

oderwat, 2018-09-05:
@dotps1 `printer.cfg` of Klipper

dotps1, 2018-09-05:
ah, ok, I will do that after this print is done so I can reload that file
thanks!

aphex008, 2018-12-27:
Adding
[gcode_macro m205]
gcode:

caused an error.

[gcode_macro m205]
gcode:
  G4

Worked fine

andresgoku, 2019-03-28:
sorry noon here 
where do you add those codes?
how can I access the gcode

dotps1, 2019-03-30:
in your printer.cfg.  you can get to it from octoprint if you have the octoklipper plugin installed.

elcojacobs, 2019-10-19:
Perhaps a better fix is to go to your printer settings in Cura and remove the M20x commands from the start G-code instead of configuring the printer to ignore them.

sfjuocekr, 2021-01-22:
Or use M400 instead

camarade35, 2022-06-12:
to disable M205, uncheck jerk control in Cura, that's all

brucetony, 2022-07-10:
> to disable M205 error message, just uncheck jerk control in Cura, that's all

This is the way

wdclub, 2022-10-09:
I have tried all:
[gcode_macro m205]
gcode:

[gcode_macro m205]
gcode:
G4

and..
[gcode_macro m205]
gcode:
G4 P0

Nothing works.. 
Unknown command:"M205" still persist :(

insippo, 2023-02-07:
M205 maybe help. Not m205