mmone/OctoprintKlipperPlugin #23: Support unicode characters in printer.cfg

pinggi, 2018-09-15:
OctoKlipper can't save `printer.cfg` containing unicode characters and effectively wipes out `printer.cfg` file after its saving.

environment:
 - Distribution version: Raspbian GNU/Linux 9
 - Octoprint version: Version 1.3.9

`
2018-09-15 21:18:09,696 - octoprint.server.api.settings - ERROR - Could not save settings for plugin OctoKlipper (0.2.5)
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 535, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 96, in on_settings_save
    f.write(data["config"])
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 410: ordinal not in range(128)
`

jameseleach, 2019-02-16:
I've got a fix for this on my branch.    Hopefully @mmone can see what I've done and integrate the changes!

Elocai, 2019-03-27:
@jameseleach could you "release" your version pls?

jameseleach, 2019-03-28:
@Elocai I can’t update the official KlipperPlugin as @mmone is the only one that can do that.  You can, however, uninstall your currently installed version and install my branch.  It’s pretty straight forward and instructions are here:
https://github.com/jameseleach/OctoprintKlipperPlugin#fork-installation-information

Basically, it’s just a matter of using PluginManager’s “From URL” option and entering this URL:
`https://github.com/jameseleach/OctoprintKlipperPlugin/archive/JEL-0.1.zip`

jass-on, 2019-06-23:
@jameseleach My printer.cfg file still gets erased using your branch.

jameseleach, 2019-07-03:
> @jameseleach My printer.cfg file still gets erased using your branch.

Can you attach your printer.cfg?  I'd like to see what encoding is used.  If it's unicode it should be converting it to utf-8 but that might not be happening.