mmone/OctoprintKlipperPlugin #30: OctoKlipper Plugin not responding with Octoprint 1.3.10rc1

creyc, 2018-11-07:
Octoprint 1.3.10 release candidate 1 is out and seems to break the OctoKlipper plugin.
https://github.com/foosel/OctoPrint/releases/tag/1.3.10rc1

Cannot connect or disconnect with klipper backend, and none of the buttons are responsive.

paeppi88, 2018-11-07:
I think this is not a problem with OctoKlipper. I disabled OctoKlipper and the Printer doesn´t connect anyway.
We should wait until 1.3.10rc2 is released.

paeppi88, 2018-11-14:
@creyc 
Try the new 1.3.10 release candidate 2. For my Printer it works with OctoKlipper again.

@mmone 
I think you can close this issue? Thank you for the great work on this plugin.

creyc, 2018-11-16:
@paeppi88 yes you're right, RC2 works again.

closed 2018-11-16 by creyc