mmone/OctoprintKlipperPlugin #32: Macros broken - triggers overtemp warning!

creyc, 2018-11-12:
I've used this simple macro to preheat the bed and hotend with one button:
```
Name: Preheat PLA
Command:
      M140 S6
      M104 S180
```
This worked fine until updating octoKlipper plugin recently. Now after the update OctoKlipper attempts to set the bed to 180C ! Fortunately the overtemp system kicks in and prevents this but this macro worked fine before the update.

oderwat, 2018-11-12:
See #19 ... But it seems that mmone is dormant.

jameseleach, 2019-02-13:
I've released a fix for this issue on my repository and I'd be interested in hearing feedback.  It's available here:  [https://github.com/jameseleach/OctoprintKlipperPlugin/releases](url)