mmone/OctoprintKlipperPlugin #33: Doesn't save config file. Just create empty printer.cfg file

ZhenyaKa, 2018-12-14:
After any editing via this plugin and press Save button I got empty cfg file!

BKLronin, 2019-01-10:
I had the same today. Saving the file via plugin worked a few times but now its just empty.

jameseleach, 2019-02-16:
I'm pretty sure this is the same bug as #23.

You can confirm by reviewing the OctoPrint log file which, by default, is here: 

`.octoprint/logs/octoprint.log`

Look for something like this:

```
2019-02-16 02:20:51,628 - octoprint.server.api.settings - ERROR - Could not save settings for plugin OctoKlipper (0.2.5)
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 541, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 109, in on_settings_save
    f.write(data["config"])
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 2486: ordinal not in range(128)
```

The character listed for you might be different.  In my example above the u'\xb0' is the degree symbol. 

The issue is with Unicode characters so I put a check in to see if they exist and then convert to a string before saving, if so.   

This resolves #23 and #33.  

I've not 'released' it yet but the code changes have been committed so you can see the fix - it's pretty simple - and you could edit your __init__.py with the change if you wanted.

jimschaefer, 2020-04-22:
Had the empty printer.cfg file problem today.  Initially working fine then pasted some settings  copied from web page and did a save.  Empty file.   Can’t seem even edit with nano now.