mmone/OctoprintKlipperPlugin #41: Plugin clean my printer.cfg

noxx2015, 2019-04-06:
hello, i think that the addon delete entries in my printer.cfg.

please look
https://community.octoprint.org/t/klipper-printer-cfg-empty-after-reboot/8773/4

thx

AlmightyFrog, 2019-12-27:
Happened to me also once now I guess. Good that i had a backup. That is anyhow and advise, as this plugins seems to not reload the config from disk once it has been loadaed in a browser-tab-session. So if you modify file both e.g. over SSH and in the plugin, plugin will not get changes which are done in the meantime without it.

sandos, 2020-04-01:
I can trigger this at will, very easy. Try inserting a german u with an umlaut: ü. I guess it will work any weird characters such as å ä ö and so on.

sandos, 2020-04-01:
This is most likely fixed in another fork: 

https://github.com/codejedi365/OctoprintKlipperPlugin/commit/31c8afcb9e63f2a42eb52bb199ceca5de2555470