mmone/OctoprintKlipperPlugin #47: Messages difficult to read w/Themeify Plugin installed

tgmorris99, 2019-07-29:
Before submitting your issue please make sure to provide the following information 

About your environment:
 - Octopi 0.16.0
 - Octoprint version - 1.3.11

When the Themeify Plugin is installed it is difficult to read the messages in the Messages area.

[image]
[octoprint.log](https://github.com/mmone/OctoprintKlipperPlugin/files/3444035/octoprint.log)

Thanks

cram42, 2019-09-11:
As a workaround for this:
- Enable customization in Themeify settings
- Expand advanced options
- Add a custom style rule (without quotes): ".log-item", "color", "#333"
[image]

tgmorris99, 2019-09-11:
Thanks. Doesn't seem to "stick" though. I make the change and it impacts any already displayed messages but new ones still show up in the old color.