mmone/OctoprintKlipperPlugin #56: Octoprint browser client keeps local cached version of printer.cfg

poorredneck, 2020-01-24:
Octoprint 1.3.12 on Raspberry Pi3b+

I have a windows 10 tablet that is dedicated to the 3D printer (computer #1). Basically just keeps Chrome tab fullscreen on Octoprint interface. I also use a laptop( computer #2) for interfacing with octoprint/klipper. 

If I access the printer.cfg editor on computer #1, save. Then open the editor on computer #2, save. 5 minutes, 1 day doesn't matter. If I open the editor on computer #1 again, it did not actually re-read the file. It kept a cached version in the browser. Make a change, save it, the changes made from computer #2 are over-written. 

If I refresh the tab on computer #1 it seems to re-read the file. So I've just made it a habit of refreshing the tab each time I want to change the config file. 

Would be nice if it re-read the file each time the editor tab is opened, if that's possible with OctoPrint's architecture.

Took longer than I'd like to admit to figure that out. I'm making alot of changes (new installation, fine tuning) and seemed like I was in a 1 step forward 2 steps back thing for a while.

ProfEngr, 2020-02-15:
I noticed this myself today.

Taudris, 2020-04-20:
This has been screwing me over with mesh leveling. I'd measure the bed, run `SAVE_CONFIG`, then make some other config change like motor current, which because of this issue would revert the mesh, leaving me wondering why the bed is ramming into the nozzle... (You have to set the Z offset past the max height of the bed, which leads to this issue when the correct mesh is not used.)