mmone/OctoprintKlipperPlugin #59: python 3.6 support

Nickjallday, 2020-04-05:
Ocotoprint will no longer support previous versions of python in the near future, it would be awesome if this plugin could get updated to be compatible with python 3.6.

Freakus, 2020-04-07:
I've been forcing compatibility for a couple weeks now and I've had no issues, though I'm not using a great deal of the available features given the hardware I have connected is a wanhao i3.

darrellenns, 2020-04-10:
@Freakus how are you forcing compatibility?

Freakus, 2020-04-10:
@darrellenns In config.yaml for octoprint:
```yaml
plugins:
    _forcedCompatible:
    - klipper
```

I don't believe the plugin manager showed it in the search, and I had to install using the "From URL" option in plugin manager using the URL shown on the plugin repository page: https://github.com/mmone/OctoPrintKlipper/archive/master.zip

Keep in mind that the _forcedCompatible option could let you load something that will break things! Don't be too surprised if you force compatibility for something and stuff goes wrong.

moi162001, 2020-04-22:
hello
adding 

plugins:
    _forcedCompatible:
    - klipper

octoprint break
an update is needed
thank you

SidShetye, 2020-05-30:
@mmone Will you be updating to Python 3?