mmone/OctoprintKlipperPlugin #66: [Feature Request] Python 3 Compatibility

definitely-not-a-t-rex, 2020-08-27:
Please add Python 3 Compatibility as outlined [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html).

cowboy3d, 2020-08-30:
Just tried to make it work with Python 3 - looks good so far but will need testing....
Dialogs show up, need to check if communication with printer works too.

damccull, 2020-09-10:
I would like to add my voice to this request. Let's get modernized. At the moment, this is the only plugin I use holding me back on 2.7.

cowboy3d, 2020-09-10:
...I have a working version here: https://github.com/cowboy3d/OctoprintKlipper3Plugin.git
For me it works now for a lot of prints for nearly two weeks - but maybe other people are using it differently then myself.
I did rename the plugin to avoid confusion / mixups with the original one... if the author decides to update his work, this would be to way to go. 
Feel free to try my version and let me know if it works for you, too.

Twilek-de, 2020-09-10:
@cowboy3d Seems to work fine here. Thanks a lot!

luco85, 2020-09-12:
Hi! For me not work. Klipper tab is ok but Plugin settings are blank, and connection tab (at sx) is missi g and i have a new tab Called octoklipper3!?😭😭

Twilek-de, 2020-09-12:
@cowboy3d Octoprint gave me the option to update Octoklipper3 to a new version (0.0.2) automatically, which totally wrecked it. I than had the same problems as @luco85. Reverting back to your original version and ignoring the update notice made things work again.

luco85, 2020-09-13:
@Twilek-de have you link of first version? Thanks

cowboy3d, 2020-09-13:
Sorry - seems I made a stupid mistake. Just corrected it and version 0.0.3 is working, at least for me.
Besides there seems to be good ones: the original plugin will get a new maintainer soon.... 
https://github.com/OctoPrint/plugins.octoprint.org/issues/599

Please let me know if you still experience problems.

Twilek-de, 2020-09-13:
Thanks a lot. I have just started a 34 hour, print so I cannot test it just yet, but I am sure it will work, I will report as soon as I can test it.

luco85, 2020-09-13:
Wow! Now it work! Thanks!!

Twilek-de, 2020-09-14:
Print failed, Octoklipper works :-)